espressif/esp-csi #201: How to increase the esp send and receive sampling rate

Yaqi3707, 2024-08-21:
Hi there,

I have a problem with the CSI sampling rate between the transmitter and receiver using the` ../examples/csi_send & csi_recv`.

I was using Rpi4 with OS: Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

My ESP-idf is: ESP-IDF v5.4-dev-1873-g41dd1a351b

From the transmitter, I set up 20000 loops for the esp_send function and increased the `#define CONFIG_SEND_FREQUENCY 50000`. I also set up system time to measure the total time cost. From the output of the terminal, it seems these 20000 packets have been sent in 15~16 seconds.
( My transmitter code [tx.txt](https://github.com/user-attachments/files/16700739/tx.txt) )

[image]

However, the receiver only captured around 200 data. I have increased the series baud rate to 3000000 in the `csi_data_read_parse.py`. Based on the captured ID in the CSV file, it seems multiple packets dropped frequently.
[image]
(Below is the CSV of my collected data)
[output18.csv](https://github.com/user-attachments/files/16700619/output18.csv)

My two transmitters and receivers are placed stationarily and right next to each other in a quiet office room. Such massive packets loss is unexpected. (By the way, I changed my esp32 to esp32s3 from my last posted issue, I want to implement the FTM example later.)
[image]

Therefore, I am concerned with two questions:
1. How can I improve the sampling rate so that my receiver can collect 800~1000 pk/s?

2. Is there any way to stabilize the data captured by the receiver? In other words, how can I make the receiver capture most of the consecutive CSI data?

Sincerely appreicate any comments and guidance.

MacChu0315-Espressif, 2024-08-22:
Hi，#135 and #173 may help, and you can try to increase the espnow rate.

Yaqi3707, 2024-08-24:
Hi there, thanks for providing the information but I have tried the approaches you mentioned in those two issues. I previously increased the serial baud rate to 3000000 in the receiver, disabled the wait function in sender, and I attempted to modify the transmission rate in the sender too.

The problem I am stuck with the sender is that I followed the recommendation to modify the transmission rate function:

```
static const uint8_t CONFIG_CSI_SEND_MAC[] = {0x1a, 0x00, 0x00, 0x00, 0x00, 0x00};
...

static void wifi_init()
{
    ...
    // ESP_ERROR_CHECK(esp_wifi_config_espnow_rate(ESP_IF_WIFI_STA, WIFI_PHY_RATE_MCS7_SGI));

    if (esp_now_init() != ESP_OK) {
        ESP_LOGE("ESP-NOW","Error initializaing ESP-NOW");
        return;
    }

    esp_now_rate_config_t rate_config = {
        .phymode = WIFI_PHY_MODE_11G,
        .rate = WIFI_PHY_RATE_MCS7_SGI,
        .ersu = false,
        .dcm = false
    };

    ESP_ERROR_CHECK(esp_now_set_peer_rate_config(CONFIG_CSI_SEND_MAC, &rate_config));
    ...
}
```
However, the system keeps crashing and rebooting after I rebuilt and reflashed. 
[image]

I tried to change the parameters of 
`.phymode = WIFI_PHY_MODE_HT40` 
and 
`.rate = WIFI_PHY_RATE_54M` referred from `esp_now.h` and `esp_wifi_types_generic.h` but none of my attempts worked.

I truly appreciate if you could provide more information and assistance on this problem.

Meanwhile, as I increased the baud rate from the receiver, why do I still receive massive "element number is not equal" and "data is incomplete"? Is this the direct problem that causes me having serious CSI data loss from the receiver side?
[image]

Yaqi3707, 2024-09-04:
Hi, @MacChu0315-Espressif Sorry for the interruption. Could you provide me more hints on how I can resolve this issue so that I can have the identical high-frequency sampling rate without any packet loss as you did in #135 ?
[image]

Thank you very much for your guidance and assistance!

MacChu0315-Espressif, 2024-09-06:
hi @Yaqi3707, just for reference and simple proof of feasibility, I only made these changes on the basis of the original procedure:
[image]
[image]
[image]
[image]

Yaqi3707, 2024-09-27:
Thank you very much. I attempted to redirect the output from the terminal but it still contained many system commands. I will try to use the SD card to write all the data into a local file as quickly as possible.

closed 2024-09-27 by Yaqi3707