espressif/esp-csi #202: ninja: build stopped: subcommand failed.

explorer1100, 2024-08-22:
C:/Users/dell/Desktop/esp-csi/examples/esp-radar/connect_rainmaker/managed_components/espressif__esp_schedule/include/esp_schedule.h:116:5: error: unknown type name 'time_t'
  116 |     time_t next_scheduled_time_utc;
      |     ^~~~~~
[1255/1302] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\dell\Desktop\esp-csi\examples\esp-radar\connect_rainmaker\build\log\idf_py_stderr_output_17852 and C:\Users\dell\Desktop\esp-csi\examples\esp-radar\connect_rainmaker\build\log\idf_py_stdout_output_17852

**File  idf_py_stdout_output_17852 last few lines  :** 
C:/Users/dell/Desktop/esp-csi/examples/esp-radar/connect_rainmaker/managed_components/espressif__esp_schedule/include/esp_schedule.h:116:5: error: unknown type name 'time_t'
  116 |     time_t next_scheduled_time_utc;
      |     ^~~~~~
[1251/1302] Building C object esp-idf/espressif__ws2812_led/CMakeFiles/__idf_espressif__ws2812_led.dir/ws2812_led.c.obj
[1252/1302] Linking C static library esp-idf\espressif__app_reset\libespressif__app_reset.a
[1253/1302] Building C object esp-idf/espressif__esp_schedule/CMakeFiles/__idf_espressif__esp_schedule.dir/src/esp_schedule.c.obj
[1254/1302] Building C object esp-idf/espressif__esp_secure_cert_mgr/CMakeFiles/__idf_espressif__esp_secure_cert_mgr.dir/srcs/esp_secure_cert_tlv_read.c.obj
[1255/1302] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.obj
ninja: build stopped: subcommand failed.

MacChu0315-Espressif, 2024-08-22:
Hello, you can try running the following commands and then repeat the operation:
```
idf.py fullclean
rm -rf build
python-clean
```
If it still doesn't work, please provide us with the complete log of the whole operation.

explorer1100, 2024-08-22:
Still facing the same error. Here is the log file : [1/1113] Generating project_elf_src_esp32.c
[2/1113] Generating ../../ota_data_initial.bin
[3/1113] Generating C:/Users/dell/Desktop/esp-csi/examples/esp-radar/connect_rainmaker/build/esp-idf/esp_system/ld/memory.ld linker script...
[4/1113] Generating C:/Users/dell/Desktop/esp-csi/examples/esp-radar/connect_rainmaker/build/esp-idf/esp_system/ld/sections.ld.in linker script...
[5/1113] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table

# Name, Type, SubType, Offset, Size, Flags

nvs,data,nvs,0xd000,32K,

fctry,data,nvs,0x15000,16K,

log_status,data,nvs,0x19000,16K,

otadata,data,ota,0x1d000,8K,

phy_init,data,phy,0x1f000,4K,

ota_0,app,ota_0,0x20000,1832K,

*******************************************************************************
[6/1113] Building C object esp-idf/mbedtls/mbedtls/3rdparty/everest/CMakeFiles/everest.dir/library/x25519.c.obj
[7/1113] Building C object esp-idf/mbedtls/mbedtls/3rdparty/everest/CMakeFiles/everest.dir/library/Hacl_Curve25519_joined.c.obj
[8/1113] Building C object esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m_driver_entrypoints.c.obj
[9/1113] Building C object esp-idf/mbedtls/mbedtls/3rdparty/everest/CMakeFiles/everest.dir/library/everest.c.obj
[10/1113] Building C object esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m/p256-m.c.obj
[11/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj
[12/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj
[13/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesce.c.obj
[14/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/asn1write.c.obj
[15/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aria.c.obj
[16/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/asn1parse.c.obj
[17/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum.c.obj
[18/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/base64.c.obj
[19/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum_mod.c.obj
[20/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum_mod_raw.c.obj
[21/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum_core.c.obj
[22/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/camellia.c.obj
[23/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/block_cipher.c.obj
[24/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/chacha20.c.obj
[25/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/chachapoly.c.obj
[26/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ccm.c.obj
[27/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/cipher.c.obj
[28/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/cipher_wrap.c.obj
[29/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/constant_time.c.obj
[30/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/des.c.obj
[31/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/cmac.c.obj
[32/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecdh.c.obj
[33/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/dhm.c.obj
[34/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecdsa.c.obj
[35/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ctr_drbg.c.obj
[36/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecjpake.c.obj
[37/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecp_curves_new.c.obj
[38/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecp.c.obj
[39/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecp_curves.c.obj
[40/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/entropy_poll.c.obj
[41/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/entropy.c.obj
[42/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/gcm.c.obj
[43/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/hkdf.c.obj
[44/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/error.c.obj
[45/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/hmac_drbg.c.obj
[46/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/lmots.c.obj
[47/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/md.c.obj
[48/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/lms.c.obj
[49/1113] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/md5.c.obj
[... cut ...]
                 from C:/Users/dell/Desktop/esp-csi/examples/esp-radar/connect_rainmaker/managed_components/espressif__esp_schedule/src/esp_schedule_nvs.c:18:
C:/Users/dell/Desktop/esp-csi/examples/esp-radar/connect_rainmaker/managed_components/espressif__esp_schedule/include/esp_schedule.h:116:5: error: unknown type name 'time_t'
  116 |     time_t next_scheduled_time_utc;
      |     ^~~~~~
[1070/1113] Linking C static library esp-idf\espressif__ws2812_led\libespressif__ws2812_led.a
[1071/1113] Linking C static library esp-idf\espressif__esp_secure_cert_mgr\libespressif__esp_secure_cert_mgr.a
ninja: build stopped: subcommand failed.

MacChu0315-Espressif, 2024-08-29:
I believe the current logs are inadequate for debugging purposes. Could you please provide the following logs?

1. From the `$IDF_PATH` directory, include logs for the following commands:
   - `git pull`
   - `git submodule update --init --recursive`
   - `./install.sh`
   - `. ./export.sh`

2. From the `PATH_TO_RAINMAKER_EXAMPLE` directory, include logs for the following commands:
   - `idf.py fullclean`
   - `rm -rf build`
   - `python-clean` 
   - `idf.py set-target esp32`
   - `idf.py build flash monitor`

Thank you for your assistance!

ilxj, 2024-11-29:
I have the same question. Any progress?

ilxj, 2024-11-29:
I solved this by adding `#include <time.h>` in `managed_components/espressif__esp_schedule/include/esp_schedule.h`