espressif/esp-csi #204: When will esp-csi support esp32c6 target ? And is there any plan to open up esp-radar soucr code ?

derricjs, 2024-08-29:
Hi~ I am working on a wifi sensing project on esp32-c6. I would like to ask when esp-csi will support esp32-c6, and if there is any plan to open up esp-radar source code

MacChu0315-Espressif, 2024-08-29:
We have now intergrated csi example of csi_send and csi_recv for C6 and other 32 chips in our internal branch. And these codes will be disclosed in near furture.
As for esp-radar, sorry that we still have no plan to make it open source right now.

derricjs, 2024-08-30:
> We have now intergrated csi example of csi_send and csi_recv for C6 and other 32 chips in our internal branch. And these codes will be disclosed in near furture. As for esp-radar, sorry that we still have no plan to make it open source right now.

Is there a specific moment in time when esp-csi and esp-radar support C6

HanRuoX, 2024-10-23:
> We have now intergrated csi example of csi_send and csi_recv for C6 and other 32 chips in our internal branch. And these codes will be disclosed in near furture. As for esp-radar, sorry that we still have no plan to make it open source right now.
Is there a specific moment in time when esp-csi and esp-radar support C6

Apbk20, 2024-12-12:
> We have now intergrated csi example of csi_send and csi_recv for C6 and other 32 chips in our internal branch. And these codes will be disclosed in near furture. As for esp-radar, sorry that we still have no plan to make it open source right now.

@MacChu0315-Espressif what is the expected timeline for this to be public?

Frostie314159, 2025-01-01:
Like the RF blobs, the radar blob is licensed under Apache, so I'll start ghidra and take a look.