espressif/esp-csi #205: console_test Problems with running python scripts that don't show the associated graph

mubai-w, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

[image]
When I finish with two slices ESP32S3 console_test instance is running python esp_csi_tool.py -p COM5 and no relevant graph appears esp_csi_tool.py where the baud rate has been changed to 115200
Or how do I use the interface of the esp_csi_tool.py with two tablets of esp32s3
The version I'm using is esp_idf v5.3

NingYuxiang, 2024-11-19:
波特率默认为2000000。

如果想要修改为115200，需要同时修改esp_csi_tool.py文件中的`baudrate`和 `console_test` 程序的`ESP_CONSOLE_UART_BAUDRATE`。并且过低的串口波特率会导致数据发送不及时导致队列溢出，出现数据丢失等情况
[image]

TindalosKorone, 2024-12-04:
### I have the same problem with him, I'm using ESP32S2, although my log is reacting, but by the time of getting CSI data, the same happens. After that, I used putty to monitor at 921600, and I could obtain CSI data, but under the same baud rate , I don't see it in console_test
[image]
- As shown in the picture, my log stops under CSI RECV

+ but you can see it in putty:
[image]

+ Even if I use another software csi_data_read_parse.py , it can be displayed:
[image]

ccjellycool, 2025-03-13:
I face the same problem. 
esp_csi_tool.py shows nothing except logs, but csi_data_read_parse.py shows a graph.

> ### I have the same problem with him, I'm using ESP32S2, although my log is reacting, but by the time of getting CSI data, the same happens. After that, I used putty to monitor at 921600, and I [...]

ccjellycool, 2025-03-17:
> I face the same problem. esp_csi_tool.py shows nothing except logs, but csi_data_read_parse.py shows a graph.

I misunderstood the esp-csi/examples/esp-radar/console_test/README.md.  
I flashed the csi_recv project, but actually, flashing the console_test project is the correct one.