espressif/esp-csi #210: 编译失败

daydayup33, 2024-10-22:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我之前使用ESP32S3可以成功编译demo console_test，现在我没有修改任何环境却编译失败，

### Actual behavior (suspected bug)

**使用5.0.2环境，直接idf.py set-target esp32s3,会有如下的报错：**
```
- > -- Building ESP-IDF components for target esp32s3
- > CMake Error at E:/WIFICSI/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/build.cmake:519 (message):
- >   ERROR: The hash sum of the component "espressif/cmake_utilities(0.5.3)"
- >   does not match the one recorded in your dependencies.lock file.  This could
- >   be due to a potential spoofing of the download server, or your lock file
- >   may have become corrupted.  Please review the lock file and verify the
- >   download server's authenticity to ensure the component's security and
- >   integrity.
- > 
- > Call Stack (most recent call first):
- >   E:/WIFICSI/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/project.cmake:447 (idf_build_process)
- >   CMakeLists.txt:10 (project)
- > 
- > 
- > -- Configuring incomplete, errors occurred!
- > See also "E:/WIFICSI/Espressif/frameworks/esp-idf-v5.0.2/examples/esp-radar/console_test5-2_chanel10/build/CMakeFiles/CMakeOutput.log".
- > cmake failed with exit code 1, output of the command is in the E:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10\build\log\idf_py_stderr_output_87560 and E:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10\build\log\idf_py_stdout_output_87560

```
**### 但是idf.py set-target esp32c3可以成功，这时候再使用命令idf.py set-target esp32s3显示正常，idf.py menuconfig也正常，无法使用idf.py build**

- E:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
-   import pkg_resources
- Executing action: all (aliases: build)
- Running ninja in directory E:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10\build
- Executing "ninja all"...
- [0/1] Re-running CMake...-- ccache will be used for faster recompilation
- -- Building ESP-IDF components for target esp32s3
- CMake Error at E:/WIFICSI/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/build.cmake:519 (message):
-   ERROR: The hash sum of the component "espressif/cmake_utilities(0.5.3)"
-   does not match the one recorded in your dependencies.lock file.  This could
-   be due to a potential spoofing of the download server, or your lock file
-   may have become corrupted.  Please review the lock file and verify the
-   download server's authenticity to ensure the component's security and
-   integrity.
- 
- Call Stack (most recent call first):
-   E:/WIFICSI/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/project.cmake:447 (idf_build_process)
-   CMakeLists.txt:10 (project)
- 
- 
- -- Configuring incomplete, errors occurred!
- See also "E:/WIFICSI/Espressif/frameworks/esp-idf-v5.0.2/examples/esp-radar/console_test5-2_chanel10/build/CMakeFiles/CMakeOutput.log".
- FAILED: build.ninja
- E:\WIFICSI\Espressif\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SE:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10 -BE:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10\build
- ninja: error: rebuilding 'build.ninja': subcommand failed
- ninja failed with exit code 1, output of the command is in the E:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10\build\log\idf_py_stderr_output_11068 and E:\WIFICSI\Espressif\frameworks\esp-idf-v5.0.2\examples\esp-radar\console_test5-2_chanel10\build\log\idf_py_stdout_output_11068

我更新为5.3.1，无法idf.py set-target esp32s3

- The following Python requirements are not satisfied:
- Requirement 'esptool~=4.8.dev3' was not met. Installed version: 4.5.1
- To install the missing packages, please run "install.bat"
- Diagnostic information:
-     IDF_PYTHON_ENV_PATH: D:\WiFiCSI\Espressif\python_env\idf5.3_py3.11_env
-     Python interpreter used: D:\WiFiCSI\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
- Constraint file: D:\WiFiCSI\Espressif\espidf.constraints.v5.3.txt
- Requirement files:
-  - D:\WiFiCSI\Espressif\frameworks\esp-idf-v5.3.1\tools\requirements\requirements.core.txt
- Python being checked: D:\WiFiCSI\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
- 
- ESP-IDF v5.3.1-dirty
请问我该如何解决问题，期待您的回复，谢谢。

### Steps to reproduce the behavior

.S

### Project release version

5.0.2 5.3.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Win11

### Shell

ZSH

NingYuxiang, 2024-11-18:
## 问题1 
在5.0.2环境中，这个错误表明 `espressif/cmake_utilities`组件的哈希值与 `dependencies.lock` 文件中记录的不匹配，可能是由于下载过程中的数据损坏或篡改，或者 `dependencies.lock` 文件本身已损坏。
### 解决方法
#### 1. 清理缓存并重新下载组件
首先尝试清理依赖缓存并强制重新下载所有依赖项：
```
idf.py fullclean
idf.py restore
```
这将清理项目的构建缓存并恢复依赖项，确保 `dependencies.lock` 文件与实际组件匹配。

#### 2. 检查 `dependencies.lock` 文件
如果上述步骤无效，可以手动检查 `dependencies.lock` 文件，位于项目根目录下。确认文件中记录的 `espressif/cmake_utilities` 版本和哈希值是否正确。你可以删除该文件，然后重新运行 `idf.py` 来生成新的 `dependencies.lock` 文件。
```
rm dependencies.lock  # 删除旧的依赖锁文件
idf.py reconfigure     # 重新配置并生成新的锁文件
```

## 问题2

在切换为5.3.1后，您的idf环境没有正确的安装，请确保您已在 esp-idf 的根目录中运行了这些命令。`git pull；git submodule update --init --recursive;  ./install.sh; . ./export.sh; `

onefishfish, 2025-02-24:
能否本地烧录，不连接RainMaker cloud ，由于烧录的时候远程下载网络不好，烧录总是不成功，本地烧录各种依赖，怎么解决 ，能提供步骤和 详细说明吗？

onefishfish, 2025-02-24:
能否增加sta-ap配网模式，增加信号自动校准人体 csi模式，不需要esp RainMaker app手动模式