espressif/esp-csi #216: 源文件打不开是什么问题

wxy834, 2024-11-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

[image]
[image]
源文件打不开

wxy834, 2024-11-20:
具体怎么操作？

NingYuxiang, 2024-11-21:
可以参考CSDN博客[解决ESP32 在 VScode环境下头文件无法打开的问题](https://blog.csdn.net/A3366_225/article/details/128234755)

glumeyukino, 2024-11-29:
> 具体怎么操作？

你好，请问解决了吗，我也是这个问题。想生成compile commands.json文件但不太会搞..