espressif/esp-csi #218: Training Radar model causes python script to crash

zathlute, 2024-12-15:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect esp_csi_tool.py to run without issue  when the radar model is trained.

### Actual behavior (suspected bug)

Instead it will crash shortly after recording the training the radar model.

### Error logs or terminal output

```shell
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: DkcQRhBED0cRRRNDFEQUQRZCF0AbQx4/HUAePR0+JD4jOyI6JTklOyM8JjsnOS44LDssOy08Kz8qPis9Jj4mQSRBIT8hPiA7ID0aOxk7FTsUOhM4DgONQs0CDIIMQQwAi4BLv8r+yg=
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v5.3.2\esp-csi\examples\esp-radar\console_test\tools\esp_csi_tool.py", line 554, in show_curve_subcarrier
    str_values = self.tableView_values[g_radio_header_pd.columns.values[j]][int(
                                                                            ^^^^
ValueError: invalid literal for int() with base 10: ''
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v5.3.2\esp-csi\examples\esp-radar\console_test\tools\esp_csi_tool.py", line 554, in show_curve_subcarrier
    str_values = self.tableView_values[g_radio_header_pd.columns.values[j]][int(
                                                                            ^^^^
ValueError: invalid literal for int() with base 10: ''
Exception: Incorrect padding, data: xDTDNsc2wjjGNsc1yjXNNM000DTON9I50TnTN9I41znYOtk12jzaOto53DveON4/3T7dPt0+3D3aP9s62DzZOdk52DbXMtox1zHXLdcr1irXJtYmiPYIdkd2hvaGNoX2hTaEtsP2wo=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: xDTDNsc2wjjGNsc1yjXNNM000DTON9I50TnTN9I41znYOtk12jzaOto53DveON4/3T7dPt0+3D3aP9s62DzZOdk52DbXMtox1zHXLdcr1irXJtYmiPYIdkd2hvaGNoX2hTaEtsP2wo=
Exception: Incorrect padding, data: QvND8ULvQ/A/7z3wPe877zztOes76jjmOOQ25DfmNOMy4jLlMd8x4DLhMN0w4C7aMtoy2zPaMtoy2jLMt4y3jPfMuI04jLlMuUv6C7oMOov7TDxLvAs8SvyJ/go9yr4JPwl/iMAIwM=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: QvND8ULvQ/A/7z3wPe877zztOes76jjmOOQ25DfmNOMy4jLlMd8x4DLhMN0w4C7aMtoy2zPaMtoy2jLMt4y3jPfMuI04jLlMuUv6C7oMOov7TDxLvAs8SvyJ/go9yr4JPwl/iMAIwM=
Exception: Incorrect padding, data: BLcDuAK5A7gCuQC8/rz9vv29/L/6vPe++L/3wfjB8cDww/LE8MTuxPDE7cPuxOrE68HpwujC7MDrwOrC7LuwvDB8sPyw/TF9cT6yPrI+8j8yv7MAMsA0AHRAtQF1AfVCNgL2gvcDt8=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: BLcDuAK5A7gCuQC8/rz9vv29/L/6vPe++L/3wfjB8cDww/LE8MTuxPDE7cPuxOrE68HpwujC7MDrwOrC7LuwvDB8sPyw/TF9cT6yPrI+8j8yv7MAMsA0AHRAtQF1AfVCNgL2gvcDt8=
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v5.3.2\esp-csi\examples\esp-radar\console_test\tools\esp_csi_tool.py", line 1024, in run
    radar_data_handle(self, series)
  File "C:\Espressif\frameworks\esp-idf-v5.3.2\esp-csi\examples\esp-radar\console_test\tools\esp_csi_tool.py", line 945, in radar_data_handle
    g_radar_status_array[-1][taget_index] = data[f"{RADAR_targetS_NAMES[taget_index]}_status"]
    ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: ''
Exception: Incorrect padding, data: uAO6A7kGuQW6B70EvgW/CL8GwAu+DMEQwBHDEMMPwhHGEsUQxhXGFcYUxRXGFMccxBrFGcQYwRfDF8MWBLCE8MSxQ/EDMYMxQzICckIygXKAswBzQHR/9L91fzU+tf42Pja9tz04PI=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: uAO6A7kGuQW6B70EvgW/CL8GwAu+DMEQwBHDEMMPwhHGEsUQxhXGFcYUxRXGFMccxBrFGcQYwRfDF8MWBLCE8MSxQ/EDMYMxQzICckIygXKAswBzQHR/9L91fzU+tf42Pja9tz04PI=
Exception: Incorrect padding, data: ugC6AbwEuQK8BL4CvwLABL8FwgjBCMENwg7DD8MNxBDFEccPyBTIE8cRxhTIE8oaxRjFGMUYxRfEGMYWxhPFE8URxw/FDckNyA3LCswJzAfOBM4D0ALSANT/1/7X/Nj72vrb+N324P=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: ugC6AbwEuQK8BL4CvwLABL8FwgjBCMENwg7DD8MNxBDFEccPyBTIE8cRxhTIE8oaxRjFGMUYxRfEGMYWxhPFE8URxw/FDckNyA3LCswJzAfOBM4D0ALSANT/1/7X/Nj72vrb+N324P=
```

### Steps to reproduce the behavior

1. C:\Espressif\frameworks\esp-idf-v5.3.2\esp-csi\examples\esp-radar\console_test\tools>python esp_csi_tool.py -p COM7
2. Under train, Delay = 10sec, Duration = 1min
3. Select start
4. Select Yes

### Project release version

latest version

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11 Home

### Shell

CMD

### Additional context

The errors in the log are persistent directly after starting the application, although the program still seems to run okay with a slight lag as time run time progresses. When you try to train the radar model it will work for around a minute before completely crashing

NingYuxiang, 2025-01-07:
I suspect that the data format output by the chip is inconsistent with the one in the esp_csi_tool.py file. Could you provide the chip's log for analysis?

zathlute, 2025-01-07:
Here is the Log Data for the chip when run on a surface 7 (Windows Related to this ticket)

[log_data.txt](https://github.com/user-attachments/files/18331281/log_data.txt)

Log from Mac (Different computer with the same issue)

NingYuxiang, 2025-01-07:
I have observed that the error logs you provided indicate missing information, which may be caused by the console_test project using a UART baud rate of 921600. This places a performance requirement on the UART-to-USB adapter, and currently, your adapter may not meet the necessary specifications.

There are two solutions for you to choose from:

#### 1. Replace with a higher-quality UART-to-USB adapter, such as one based on the CP2102 chip.
#### 2. Switch to using usb_serial for data transmission:
- In `app_main.c`, modify the following code:
```c
    esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
    repl_config.prompt = "csi>";
    ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
```
```C
    esp_console_dev_usb_serial_jtag_config_t hw_config = ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
    repl_config.prompt = "csi>";
    ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&hw_config, &repl_config, &repl));
```
- In menuconfig, select ESP_CONSOLE_USB_SERIAL_JTAG.
- Connect the hardware board using usb_serial.

zathlute, 2025-01-07:
The board that I'm using has the UART chip CH340c, looking at the data sheet it should be able to provide a baud rate of 921600. I've downloaded the latest driver for that specific and that fixed the issues I had with flashing at 921600. 

I managed to test it on my Mac, although i do no know if that was sufficient as there is still issues accessing bin_data. This output is continuous while the python app is running see terminal output below:

json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: wSCFaogtRGmHJoVixF/FXsNcBhoBVgBUQFNBU39Lvkq9iYBF/kb8hfyB+oD5vXe/dfxy+W/+b/xtO28ALT9sP2wA7QHqAO4E6wevCa8MrwyyzrDPc9J01HeWPJc8nD9dxF/EYskljc=
Exception: Incorrect padding, data: O8YvviPOJ74T1gfO/drt3u3e4erN3r3eufKx7q3ulfKB8oUCcfpt/mn6WfpR/kECOfYm8hb2KuoS3hrqItYa0iLSJs4mvirWJrpJxk2+TbZhvl2ybcJ1uoHClc6dwrXGxc7V0uX/eI=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: O8YvviPOJ74T1gfO/drt3u3e4erN3r3eufKx7q3ulfKB8oUCcfpt/mn6WfpR/kECOfYm8hb2KuoS3hrqItYa0iLSJs4mvirWJrpJxk2+TbZhvl2ybcJ1uoHClc6dwrXGxc7V0uX/eI=
Exception: Incorrect padding, data: 0nHPccddw1nDWbdVp12fUZ9Zi0WHPXc5z1vOXM1XypTKlsqTxZLHkcSSxY7Di8CN/4n8j/uL94v0jvaQ9JL0kfOU9Jb1lfWX9Jr4m/ic+p/9oPyiwKTDo8WmyafLqs1r0m3Ubdlv3U=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: 0nHPccddw1nDWbdVp12fUZ9Zi0WHPXc5z1vOXM1XypTKlsqTxZLHkcSSxY7Di8CN/4n8j/uL94v0jvaQ9JL0kfOU9Jb1lfWX9Jr4m/ic+p/9oPyiwKTDo8WmyafLqs1r0m3Ubdlv3U=
Exception: Incorrect padding, data: mvei5KLYotimzbHFub29scWuwZ7Qmty3IPYe9x3+2zzXPNc80n/TQNJCjoSPRYqIi41LkEiQSpBIlAiRDJQMlAyTEFYSVRFVFE9WDlcNWg1bDFsKXcefxaLA48Dku+W5KbcqsWxsbk=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: mvei5KLYotimzbHFub29scWuwZ7Qmty3IPYe9x3+2zzXPNc80n/TQNJCjoSPRYqIi41LkEiQSpBIlAiRDJQMlAyTEFYSVRFVFE9WDlcNWg1bDFsKXcefxaLA48Dku+W5KbcqsWxsbk=
Exception: Incorrect padding, data: dft985D7if+Y/6j3s/+798P7yvvW/+UD8QP1A/X/Fwoe/yIDMgcyDi4SRQ5JElUWVSVgInQtbDlwOXdaEFsSGhQWExcUFxMVVQ9UDZQLkgyRCpEG0wPPQMx/SnpJeIlyhrDErP/pPM=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: dft985D7if+Y/6j3s/+798P7yvvW/+UD8QP1A/X/Fwoe/yIDMgcyDi4SRQ5JElUWVSVgInQtbDlwOXdaEFsSGhQWExcUFxMVVQ9UDZQLkgyRCpEG0wPPQMx/SnpJeIlyhrDErP/pPM=
Exception: Incorrect padding, data: D0ALRA9dB10CaQJxAn36jvaS/qbuouq77sfqx+7T5uDi7OLw5gjeDNkQ0RzJINE1vTGsPZxFpEGYQZRBnDWMPZA1lDV/IIwkgxSYDJQEkACY7JTsnOCgz6zHuce5s8mq2ZrmvZ5Am0=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: D0ALRA9dB10CaQJxAn36jvaS/qbuouq77sfqx+7T5uDi7OLw5gjeDNkQ0RzJINE1vTGsPZxFpEGYQZRBnDWMPZA1lDV/IIwkgxSYDJQEkACY7JTsnOCgz6zHuce5s8mq2ZrmvZ5Am0=
Exception: Incorrect padding, data: 7B4AKwQWCB4QEiAWJAY5AjUGRfVWAmb5ZFq7XL1i+mX5Y/Zp92j3azhuN3A0NDM3d3t3f7h5ekC8fnh8fn5/vn+7QLtDu358Q7QCtASzBq0Grwepw6jHpcahxJ/FmIaVhJJGjUOIA4=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: 7B4AKwQWCB4QEiAWJAY5AjUGRfVWAmb5ZFq7XL1i+mX5Y/Zp92j3azhuN3A0NDM3d3t3f7h5ekC8fnh8fn5/vn+7QLtDu358Q7QCtASzBq0Grwepw6jHpcahxJ/FmIaVhJJGjUOIA4=
Exception: Incorrect padding, data: AlT1SO1A6TjQONA40DC4MLQrrB+nF5MPjwd/An8Hau5e7mLyWuJN2VHZRc1FzTG9LbEtrCSgLZgokC2MOYg1gz1/RYhWg1GMVohmmHKUcqB/rHqsi7GXxZvJo96s6rjuvALMC8wrTw=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: AlT1SO1A6TjQONA40DC4MLQrrB+nF5MPjwd/An8Hau5e7mLyWuJN2VHZRc1FzTG9LbEtrCSgLZgokC2MOYg1gz1/RYhWg1GMVohmmHKUcqB/rHqsi7GXxZvJo96s6rjuvALMC8wrTw=
Exception: Incorrect padding, data: gyeLGpMKlxKX/qf+tO205cDduNDM1d3E1bjdsN2nAp8CjwKPCoMWdhJ2J24rbitiO2pPZmRWWGJgXmheYHJsamxqZHZwg2yDbI9Lm0ufT5tHp0enN6wruCe4DsAKyPnM4dnV4cDp/E=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: gyeLGpMKlxKX/qf+tO205cDduNDM1d3E1bjdsN2nAp8CjwKPCoMWdhJ2J24rbitiO2pPZmRWWGJgXmheYHJsamxqZHZwg2yDbI9Lm0ufT5tHp0enN6wruCe4DsAKyPnM4dnV4cDp/E=
Exception: Incorrect padding, data: PBc8IA0NDgrPCtIG1AbXRdpF3kLcQKGC5YHmg+a6qbyu+639s/q0+rP3uTR4NHwwfC18LH4qPSo+JT4qOyc8JzsoOSM05TckMukw6y7tbOsprWmsZrFlsmK3obiee51/mUPWR9UOEg=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: PBc8IA0NDgrPCtIG1AbXRdpF3kLcQKGC5YHmg+a6qbyu+639s/q0+rP3uTR4NHwwfC18LH4qPSo+JT4qOyc8JzsoOSM05TckMukw6y7tbOsprWmsZrFlsmK3obiee51/mUPWR9UOEg=
Exception: Incorrect padding, data: wOXI2dDI1cjQvNW83bThn+mb6Yf1i/56+W7+ZgJeDlEOQQpBEi0aKB4oKxgvIDMIQxBUDFwAYARkAGgEZAxwDHAMcBR4KGwkdDFkPWhYEFUWlxeT1pDbjt2K38ihxqTBpv+rOW00MA=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: wOXI2dDI1cjQvNW83bThn+mb6Yf1i/56+W7+ZgJeDlEOQQpBEi0aKB4oKxgvIDMIQxBUDFwAYARkAGgEZAxwDHAMcBR4KGwkdDFkPWhYEFUWlxeT1pDbjt2K38ihxqTBpv+rOW00MA=
Exception: Invalid base64-encoded string: number of data characters (137) cannot be 1 more than a multiple of 4, data: 4j3eSe5g5ljyZP1gBXQVdw17HIcViySeKKsqjCqN7lHvUPFUtBS2FrYUuhe5Fr3XgdS/14KSyJWGk8eTxZDIkMiOxYwFjsDJA4o+xz3JPMV3BzgFdQRyRi9EaYVogWSCYcFdAlBUU=
json.JSONDecodeError: cannot access local variable 'bin_data' where it is not associated with a value, data: 4j3eSe5g5ljyZP1gBXQVdw17HIcViySeKKsqjCqN7lHvUPFUtBS2FrYUuhe5Fr3XgdS/14KSyJWGk8eTxZDIkMiOxYwFjsDJA4o+xz3JPMV3BzgFdQRyRi9EaYVogWSCYcFdAlBUU=
[... cut ...] BDqAPH6++z45gLcD84TxCC6Lak=
Josephs-iMac:tools zathlute$ /usr/local/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/multiprocessing/resource_tracker.py:276: UserWarning: resource_tracker: There appear to be 6 leaked semaphore objects to clean up at shutdown: {'/mp-g55lzi0h', '/mp-zxydrvtw', '/mp-qs1ifxb3', '/mp-f0yfzs23', '/mp-jwbkd762', '/mp-34rkbg0_'}
  warnings.warn(

I will try switching to using usb_serial for data transmission tomorrow when i get a chance

NingYuxiang, 2025-01-08:
Your output is continuous, but characters are missing, such as a comma missing in the yellow circle in the image, causing a parsing error in the Python application.
[image]

zathlute, 2025-01-08:
Using the usb_serial for data transmission option works great! Thankyou for your help.

I have one more question, I'm planning on basing my university final year project on using CSI data to track movement. My supervisors recommended that using two antennas on the receiver would enable much more accurate tracking as the phase shift between the antennas would allow for pin pointing locations. From what i can find there is no way to have more than one antenna on the ESP32, Do you have any advice on how to get around this limitation? if not i may need to research alternative hardware.

NingYuxiang, 2025-01-08:
Thank you very much for your feedback. I'm glad to hear that you've had good results using usb_serial for data transmission!

Regarding the antenna phase synchronization issue, we have a multi-chip RF phase synchronization solution. If you're interested in further communication, could you please share your email address? We look forward to engaging in a deeper discussion and providing you with support.

zathlute, 2025-01-08:
Sure, you can contact me at zathlute@gmail.com

NingYuxiang, 2025-01-13:
Sure, I have already contacted you via email.