espressif/esp-csi #230: ESP32C5发送端发送了数据，但CRAD并未接收

Duanyy0813, 2025-07-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

我在使用单发双收模式的开发板，原始自带的程序可以直接使用，但我重新烧录了发送端的程序，接收端程序未改动，虽然显示发送了CSI数据，但接收端未打印相关接收数据，是否是需要配置些额外的参数？

NingYuxiang, 2025-07-18:
大概率是因为预烧录的固件和目前开源的固件不是同一个信道，请按照发送端的信道配置给接收端也重新烧录

Duanyy0813, 2025-07-19:
> 大概率是因为预烧录的固件和目前开源的固件不是同一个信道，请按照发送端的信道配置给接收端也重新烧录

您好，谢谢您的回复，我重新烧录了两端的程序，不过似乎还是无法直接工作。我发现，当我为发送端供电时，接收端CARD的LED开始闪烁，并新打印出以下信息：
W (14377) wifi_rx_gain: agc_gain: 29, fft_gain: 15
W (15697) wifi_rx_gain: agc_gain: 29, fft_gain: 16
看起来接受天线开始工作了，不过CARD屏幕上依然没有CSI波形，您说的信道配置应该在那部分修改呢？